ThunderFly-aerospace/TFSIK01 #1: Optimize SiK modem layout

kaklik, 2020-10-31:
Shrink the modem layout, but maintaining conditions from [AN791](https://www.silabs.com/documents/public/application-notes/AN791.pdf)

kaklik, 2021-05-26:
Aktuální návrh nějak výrazně ignoruje doporučení z výše zmíněného application note. 

Kde jejich doporučený layout vypadá takhle: 

[image]

Pro náš případ na rychlý pohled vidím že minimálně: 

1. LC matching network se vůbec nesnaží být blízko IC
2. Napájení TX části je až několik desítek milimetrů od IC. 
3. Není moc izolován přeslech mezi RX a TX vylitou zemí mezi nimi. 

A pak také jsou v návrhu některé základní obecné problémy, s layoutem RF cest.  Například není možné tahat jakoukoliv cestu paralelně na protější straně RF cesty. 

[image]

Taktéž musí být maximálně minimalizovány přerušení protilehlé GND pod RF cestou. 

[image]

Lze asi akceptovat přerušení GND pod součástkou, kde už tak jako tak dochází k nespojitosti pole.

kaklik, 2021-05-27:
Tohle se mi moc nelíbí: 

[image]

Znamená to, mimo jiné, že výstupní RF konektory jsou prakticky úplně izolovány od země ve druhé vrstvě. 
Druhý důsledek se kterým může být problém je to, že ty cesty takhle budou fungovat jako anténa a bude na nich vysílaný signál, který bude nějak ovlivňovat signál, který na těch ovládacích signálech má být jen on/off.

Jako řešení bych navrhoval nějak lépe proházet IO piny a jejich signály. 
A cesty vést z levé strany, nikoliv z obou stran a pak udělat prostup pod L5 a L7, případně L3. 
V případě že všechno nejde přeházet tak, aby to bylo jen zespodu, tak jeden signál může vést i pod C12 a L9. 

Zkrátit by to chtělo i napájení. Protože napájení je z RF pohledu taky signál.

ChroustJan, 2021-06-04:
@kaklik  provedl jsem editaci. Prosím o kontrolu.

kaklik, 2021-06-04:
Super! Tohle řešení je mnohem lepší! 

[image]

Potíž vidím v tom, že indikační LED D1, která indikuje přepínání RF RX/TX. Protože kdyby na modulu mělo být RF stínění, tak bude vypadat asi nějak takhle: 

[image]

Tím pak nastane situace že na tuto LED nebude vidět.  Řešení by bylo buď LED přemístit k ostatním mimo, RF stínění, což je ale komplikováno přerušením GND od krystalu a místem.  Nebo LED uzpůsobit tak, aby svítala skrz PCB (To by ale pak bylo potřeba i u těch ostatních). Případně tuto LED úplně zrušit.

ChroustJan, 2021-06-07:
Navrhuji LED zrušit.

kaklik, 2021-06-07:
> Navrhuji LED zrušit.

OK..

kaklik, 2021-07-19:
Aktuálně je ještě potřeba: 

- [ ] dořešit nepříjemnost s indikační LED pod stíněním. 
- [ ] Připravit nějaké plochy, kde bude připojena stínící Cu páska
- [ ] Vyexportovat obrys do SVG, nebo DXF, aby se z toho dal v openscadu nakreslit 3D tisknutelný objekt. 
- [ ] Zarovnat vias

ChroustJan, 2021-11-18:
Dodán chybějící popisek a vygenerování 3D modelu ve formátu .STEP. Zarovnání vias jsem nějaké udělal. Musí dodělat ten kdo to tu psal, aby dořešil ty zbylé co mu přišli nezarovnané.

kaklik, 2021-11-18:
> Dodán chybějící popisek a vygenerování 3D modelu ve formátu .STEP. Zarovnání vias jsem nějaké udělal. Musí dodělat ten kdo to tu psal, aby dořešil ty zbylé co mu přišli nezarovnané.

Ani jeden z těchto kroků ale neřeší výše zmíněné body.  Na dodání popisku antén je navíc jiné issue #12. STEP model nepomůže k tomu, aby šla v openscadu udělat kostra pod stínění.. 
Zarovnání vias asi můžu dořešit.

ChroustJan, 2021-11-18:
@kaklik takovéto SVG jsi si představoval? https://github.com/ThunderFly-aerospace/TFSIK01/blob/TFSIK01A/hw/sch_pcb/TFSIK01A-Edge_Cuts.svg

kaklik, 2021-11-18:
Porovnal jsem vias i potisky, tak aby se vzájemně nepřekrývaly. 

[image]

Některé potisky jsem rovnou přesunul i tak, aby nebyly pod součástkami, nebo na ploše určené pro letování.

[image]

Zároveň jsem si ale všiml, že DRC u tohoto PCB nevypadá moc dobře: 

[image]

Největší potíž jsou myslím chyby typu "Track has unconnected end", která obvykle je způsobená tím, že na cestě jsou položeny ještě další nikam nepřipojené segmenty.

kaklik, 2022-01-09:
Pro výrobu jsou problém následující body: 

- Není jasné, kam zmizela šedivá vrstva, označující stíněnou RF část. S tím souvisí i záležitost, jak má vypadat ten separační výtisk, který bude dělat podporu stínící Cu pásce. Která v tomto řešení má nahrazovat stínící krabičku. 
- Jestli při ztrátě vrstvy nedošlo ke ztrátě i dalších změn v návrhu
- Není také jasné, jak se vypořádat s DRC chybami, kterých je PCB nyní plné.

closed 2025-08-14 by kaklik