ThunderFly-aerospace/TFSIK01 #3: Vypínač sériové linky pro USB převodník

roman-dvorak, 2020-10-31:
Holybro modemy jako USB převodník využivají FTDI čip [FT230XS](https://www.ftdichip.com/Support/Documents/DataSheets/ICs/DS_FT230X.pdf). Před ním mají "vypínač" z logických členů [HC126](https://www.ti.com/lit/ds/symlink/sn74hc126.pdf) (přesnou součástku jsem nedohledal jedná se však o tuto řadu), který sériovou linku modemu s FTDI převodníkem propojí po připojení napájení z USB. 

Schéma zapojení přepínače je na následjícím obrázku.
[image]

ChroustJan, 2021-04-25:
Možný typ přepínače: https://www.ti.com/lit/ds/symlink/ts3a44159.pdf?ts=1619372648476&ref_url=https%253A%252F%252Fwww.ti.com%252Fproduct%252FTS3A44159

kaklik, 2021-05-13:
Není lepší sem místo těch několika tranzistorů dát prostě invertor s 5V tolerantním vstupem? 

[image]

Například tohle zřejmě zároveň nahradí i ty tranzistory u diod [MC74LCX06](https://www.onsemi.com/pdf/datasheet/mc74lcx06-d.pdf)

ChroustJan, 2021-05-13:
@kaklik nápad je to zajímavé. Ale pokud to dobře čtu, může to mít i 0,5 V nízký stav to mi přijde docela vysoko.

kaklik, 2021-05-13:
> @kaklik nápad je to zajímavé. Ale pokud to dobře čtu, může to mít i 0,5 V nízký stav to mi přijde docela vysoko.

Ano, samozřejmě.. To není překvapivé, je to vlastnost bipolárních tranzistorů. Například [tranzistor BC817](https://www.onsemi.com/pdf/datasheet/bc817-40w-d.pdf), který je tam osazený má tohle napětí až 0.7V a 0.5V je běžná hodnota.

ChroustJan, 2021-05-14:
Pokud to dobře chápu tak to napětí je závislé na proudech atd... A umí být u BC817 menší.

kaklik, 2021-05-14:
> Pokud to dobře chápu tak to napětí je závislé na proudech atd... A umí být u BC817 menší.
> 

Ano, stejně jako umí být menší u toho klopného obvodu..  Nerozumím tomu co tu řešíme.. je to fundamentální omezení, které je u obou součástek stejné.  Pro tu aplikaci 0.5V nebo 0.8 V nebo i 1V prakticky nehraje roli..    
Moje pointa byla v tom ušetřit počet součástek a tím pádem v důsledku i plochu na PCB.

kaklik, 2021-05-25:
Nakonec je návrh bez USB, takže toto issue je tím vyřešeno.

closed 2021-05-25 by kaklik