ThunderFly-aerospace/TFSIK01 #6: Power voltage rise time control

kaklik, 2021-04-26:
U starších modemů založených na Si1000 byl pozorován jev, kdy při pomalém náběhu napájení se modem správně neinicializuje.
@roman-dvorak  od toho možná někde má i nějaké grafy. 
Je potřeba si dát pozor, aby stejný problém nemohl vzniknout i v nové konstrukci.

ChroustJan, 2021-04-29:
@kaklik jak probíhá pomalý náběh napájení?

kaklik, 2021-04-29:
> @kaklik jak probíhá pomalý náběh napájení?

Některé měniče, například ten ve VESC, udělá pomalý náběh napětí při zapnutí.. Trochu doufám, že k tomu má @roman-dvorak  nějaký snímek z osciloskopu, protože já nevím kde je.

ChroustJan, 2021-04-29:
Tohle nevím, jak moc dobře řešit. Jedině, že by jsme hlídali úroveň napětí a dokud nebude větší tak to nezapneme. Ten náš stabilizátor má EN pin.

kaklik, 2021-04-29:
> Tohle nevím, jak moc dobře řešit. Jedině, že by jsme hlídali úroveň napětí a dokud nebude větší tak to nezapneme. Ten náš stabilizátor má EN pin.

ano jsou v zásadě dvě možnosti jak to udělat. Buď povolovat napájení ze stabilizátoru, po tom co je stabilní a nebo držet reset dokud se neustabilizuje napájení a pak ho pustit. 
Některý měniče mají právě z tohoto důvodu PG pin, který jde použít na uvolnění resetu po tom, co se stabilizuje napájení.

roman-dvorak, 2021-04-29:
> > @kaklik jak probíhá pomalý náběh napájení?
> 
> Některé měniče, například ten ve VESC, udělá pomalý náběh napětí při zapnutí.. Trochu doufám, že k tomu má @roman-dvorak nějaký snímek z osciloskopu, protože já nevím kde je.

Tak je to přesně naopak. Viz tento zápis: 
[image]

kaklik, 2021-04-29:
Zajímavé. Možná tak není problém s rychlostí náběhu, ale spíše s tím zákmitem na začátku.   V [datasheetu](https://www.silabs.com/documents/public/data-sheets/Si106x-8x.pdf) jsou následující podmínky, kdy vznikne reset: 

[image]

Aby to spolehlivě fungovalo, tak je zřejmě potřeba se všem vyhnout.

kaklik, 2022-11-06:
Tenhle problém souvisí s náběhem napájení zřejmě jen okrajově. Reálnou příčinou je že se uzeměním CTS aktivuje bootloader.  Si10XX ale nemá dostatečně silný pull-up, který by zajistil log 1 při náběhu napájení.  Místo toho, má naopak logiku, která při log nule na vstupu pull-up vypne. V autopilotovi navíc je obousměrný translator, který se podle logických stavů snaží odhadnout směr komunikace.  Kombinace těchto dvou řešení způsobí, že se při spouštění zařízení prochází nedefinovanými stavy. Výsledkem tak je, že spuštění nebo zaseknutí v bootloaderu závisí na rychlosti náběhu napájení. 

Korektním řešením by možná bylo přidání externího pull-upu na signál CTS.

closed 2025-09-08 by kaklik