ThunderFly-aerospace/TFSIK01 #8: Last time buy of LNA IC

kaklik, 2021-05-25:
The selected chip [SPF5189Z](https://www.qorvo.com/products/p/SPF5189Z#parameters) has End of Life announced on March 8, 2021.
The manufacturer proposes the use of [QPL9547](https://www.qorvo.com/products/p/QPL9547#parameters), which has a similar characteristic on the first look. 

The noticeable feature is Shutdown input, which allows direct control of LNA from MCU IO pin. Therefore the current LNA control logic could be significantly reduced: 

[image]

To the following circuit

[image]

The unknown value is a value of resistor R4, which sets the bias point. Datasheets do not explain the possible range of values. But it is suspected, that current thought this resistor should be in the range from 65mA to 100mA. 

----

@ChroustJan  Dovedeš osadit i tu novou verzi?

ChroustJan, 2021-05-27:
Ano. @kaklik takže takto má vypadat nové výsledné zapojení? 
[image]

R4 zvoleno 3k32 po vzoru Evaluation Board – QPL9547EVB-01

kaklik, 2021-05-27:
Na tom GPIO2 bych zachoval tu LED D5, co tam vlastně byla.   Dal bych k ní popisek RF RX, pokud bude z napájení do pinu, tak bude svítit, když je zapnuný LNA (ten je zapnutý v nule).

Jinak je to ok. .. Ta  hodnota toho R4 se asi bohužel bude muset experimentálně vyzkoušet, na to řešení teď nemám.

ChroustJan, 2021-06-04:
Aplikován QPL9547 a přidána LED.

closed 2021-06-04 by ChroustJan