ThunderFly-aerospace/TFSIK01 #9: Možnost zasouvání PCB do drážek držáku

kaklik, 2021-05-25:
Při plánování rozměrů PCB jsme předpokládali, že plošný spoj půjde zasouvat do drážek ve kterých se pak pojistí šrouby zašroubovanými v místech výřezů do desky. 
Předpokládá to ovšem, že drážka může mít šířku alespoň srovnatelnou s hloubkou zubu pro šroub na boku PCB. Tomu ale brání aktuální pozice některých součástek na PCB. 

Zejména krystalu Y2, dobře na tom ale nejsou ani LED D2 a D3.  

[image]

U krystalu je tato záležitost ale obvzláště zřejmá, neboť je krystal navíc i výrazně daleko od samotného IC. 
Podobně je na tom kondenzátor C1 na protilehlé straně. 

[image]

Domnívám se, že umístění součástek by mělo být takové, aby pro uchycení PCB byla k dispozici alespoň šířka odpovídající hloubce půlkulatých výřezů na stranách PCB.

ChroustJan, 2021-05-26:
Ale ten půlkulatý výřez má 1,5mm.

ChroustJan, 2021-06-04:
Je splněn požadavek na minimální mezeru 1mm od okraje.

closed 2021-06-04 by ChroustJan

kaklik, 2021-06-04:
> Je splněn požadavek na minimální mezeru 1mm od okraje.

No já nevim, jestli je jako dobrý tohle prostě takhle zazdít, s tím že je "splněno", něco co jsme jen odhadli.  Když to zároveň není nikde namontované..  Myslím že nějaká krabička se bude dělat o dost hůře v případě, že tam bude jen tak tak 1mm.