ThunderFly-aerospace/TFSIK01 #15: RX/TX control

kaklik, 2022-09-13:
Vypadá to, že při návrhu modulu, došlo k odlišnosti od původního [RFSWITCH01A](https://github.com/mlab-modules/RFSWITCH01). Ten měl původně RX a TX ovládané pouze jedním pinem, který zároveň vypínal LNA (ve schéma SW1_CTRL). 

[image]

U nového modulu ovšem došlo k rozdělení tohoto ovládání na dva piny

[image]

Důsledkem je, že nelze použít původní firmware pro pozemní stanici tak jak je, neboť ten předpokládá ovládání pouze přes jeden pin. 

Vůbec jsem nevěděl že k této změně došlo. Můžete se prosím @povik  a @ChroustJan  domluvit jak tuto situaci vyřešit?   Já bych považoval za přirozené, aby ve firmware vznikla nová definice desky TFSIK01A.

kaklik, 2022-09-13:
Dále jsem zjistil, že nový Holybro modem předpokládá také ovládání dvěma piny, ale jsou řízeny komplementárně.

[image]

Je tak pravděpodobné, že upstream firmware je pro řízení RX/TX dvěma piny připraven. Problematická je akorát jejich polarita, která mi pro náš případ vychází stejná u obou signálů.

povik, 2022-09-13:
> Můžete se prosím @povik a @ChroustJan domluvit jak tuto situaci vyřešit? Já bych považoval za přirozené, aby ve firmware vznikla nová definice desky TFSIK01A.

Můžeme taky jen rozšířit/upravit definici té současné, pokud půjde. To bych zkusil prvně. Jinak ze starých emailů plyne, že to byla záměrná změna k usnadnění návrhu.

> Problematická je akorát jejich polarita, která mi pro náš případ vychází stejná u obou signálů.

Vychází mi to komplementárně i v našem případě. Totiž vůči RFSWITCH se prohodil smysl RF1 a RF2 u RX/TX přepínače. Nebo se U3 zapíná v nule?

kaklik, 2022-09-13:
> Vychází mi to komplementárně i v našem případě. Totiž vůči RFSWITCH se prohodil smysl RF1 a RF2 u RX/TX přepínače. Nebo se U3 zapíná v nule?

U3 se zapíná v nule, viz [datasheet](https://www.qorvo.com/products/d/da007268).

kaklik, 2022-11-06:
@povik to vyřešil a je i vyzkoušeno proti jednomu Holybro modemu z vírníku, že to funguje. Ve [firmware](https://github.com/ThunderFly-aerospace/SiK), ale chybí commit řešením.

closed 2023-08-22 by kaklik